AsteroidOS/asteroid #234: [Feature Request] Notification improvements with a phone

VasilisThePikachu, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
Another suggestion taken from WearOS about how notifications from a phone can be handled.

**Describe the feature you'd like**
When a notification is dismissed on the watch, it should also be dismissed on the phone to mirror how WearOS handles it
Also when dismissing one notification, you should still stay in the notifications panel instead of being kicked back to the watch face

**Additional context**
None yet

sevmonster, 2023-02-28:
> More icons/Take icons from phone (Snapchat seems to be the only one showing an icon)

For new Android Conversation type notifications, this could be the user picture. For media streams, it could be the album art. Some other apps provide custom icons but not many. Fallback could be app icon, instead of envelope which is often not indicative of the actual purpose of the notification.

Isakovsky92, 2026-01-11:
I will donate 100 Euro right now if develepers can say they will implement the 2-way notification syncronization feature for beluga devices within 6 months. 

Is this possible developers?