AsteroidOS/asteroid #235: fix nightly repository

beroset, 2023-01-26:
The server that produces the nightly repository does not appear to be updating everything, causing errors when trying to update using `opkg`

**To reproduce**
Steps to reproduce the behavior:
1. ssh to the watch as `root`
2. set the correct time and date 
3. establish an IP route to the internet
4. type `opkg update`
5. observe multiple failures to download with message ending in `HTTP response code said error.`

**Expected behavior**
What should happen is that the packages should all be downloaded without error

**Screenshots**
Here is an upgrade attempt on `sturgeon`:
```
root@sturgeon:~# opkg update
Downloading https://release.asteroidos.org/nightlies/ipk/all/Packages.gz.
Updated source 'uri-all-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/anthias/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/anthias/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/armv7vehf-neon/Packages.gz.
Updated source 'uri-armv7vehf-neon-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/bass/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/bass/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/core2-32/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/core2-32/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/dory/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/dory/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/lenok/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/lenok/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/qemux86/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/qemux86/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/sparrow/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/sparrow/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/sprat/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/sprat/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/sturgeon/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/sturgeon/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/swift/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/swift/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/tetra/Packages.gz.
 * opkg_validate_cached_file: Failed to download https://release.asteroidos.org/nightlies/ipk/tetra/Packages.gz headers: HTTP response code said error.
Downloading https://release.asteroidos.org/nightlies/ipk/wren/Packages.gz.
Updated source 'uri-wren-0'.
```

**Device (please complete the following information):**
 - Watch Codename: all except `wren`
 - AsteroidOS Builddate 20230108190811
 - Version: built from source

**Additional context**
If we attempt to verify using a browser or `wget` from a computer the server returns a 404 error:
```
wget https://release.asteroidos.org/nightlies/ipk/sturgeon/
--2023-01-26 06:58:18--  https://release.asteroidos.org/nightlies/ipk/sturgeon/
Resolving release.asteroidos.org (release.asteroidos.org)... 2a01:4f8:192:63d0::2, 144.76.113.209
Connecting to release.asteroidos.org (release.asteroidos.org)|2a01:4f8:192:63d0::2|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-01-26 06:58:19 ERROR 404: Not Found.
```

It apperars that something has changed so that only the *last* watch built (`wren` in alphabetical order) updates the site.  The problem has been occurring since at least 14 January 2023.

MagneFire, 2023-01-29:
All files should are now back: https://release.asteroidos.org/nightlies/ipk/ I'm not sure if the issue is now resolved though.

beroset, 2023-02-09:
Packages from asteroid-community no longer seem to be appearing on the nightlies web site, so packages such as `asteroid-qmltester` and also the qemux86 version of the SDK (which should be here: https://release.asteroidos.org/nightlies/sdk/oecore-x86_64-core2-32-toolchain-nodistro.0.sh ) are missing.

beroset, 2023-09-19:
This was partly fixed by @MagneFire and `asteroid-community` projects are now appearing.  I will close this and open a new issue for the last remaining item, which is the missing qemux86 toolchain installer.

closed 2023-09-19 by beroset