AsteroidOS/asteroid #239: [catshark/boot loop] Unable to boot after full installation

AcedTimo, 2023-02-14:
**Describe the bug**
So, after following the instructions for the Catfish/ Catshark models, my watch gets stuck in a boot loop.
The strange thing is that i was able to run it perfectly fine via the Temp Install method
As in, copying the image over and booting from the fastboot file on my pc.
However when i flashed both files to the watch, it no longer seems to be able to boot.
One more note, it gets stuck on i would assume the first frame of the AsteroidOS boot animation.
Any help in resolving this would be greatly appreciated.

**To reproduce**
Steps to reproduce the behavior:
1. Go to https://asteroidos.org/install/catfish/
2. Follow "Real Installation" instructions
3. Attempt to boot the watch.

**Expected behavior**
Well, fairly obvious but it was expected that the watch would boot into AsteroidOS

**Device :**
 - Watch Codename: catshark
 - AsteroidOS Builddate: Unsure (Latest from the installation guide)
 - Version: nightly

eLtMosen, 2023-02-14:
Hello and welcome @AcedTimo 
The first solution to try is just to flash again and try to reflash up to 4 times until it worked.
Connection via pogopins is not always perfect and failed flashes while fastboot shows no error are not uncommon.
You could try to clean the contact before flashing again, ideally with a alcohol and a microfiber cloth or toothbrush.
Or even fixiate the watch to the cradle with rubber bands or some temporary ductape.

Since the temporary install worked, the full flash install should too.

AcedTimo, 2023-02-14:
Thanks a lot for your quick response.
I had already tried flashing it multiple times.
Now i made sure to clean the contacts, on top of that i went and used a linux system to flash the watch.
Just in case Windows was being a pain again as usual.
Turns out, it worked on the first attempt.
Weather it was the contacts or Windows/ a messed up ADB installation, im not sure.
But since its resolved now, ill go ahead and close the issue.
Once again, thanks a lot.

closed 2023-02-14 by AcedTimo