AsteroidOS/asteroid #243: analog-weather-glow watchface no longer works

beroset, 2023-03-11:
**Describe the bug**
The `analog-weather-glow` watchface used to work on my `catfish` but it no longer does.  

**To reproduce**
Steps to reproduce the behavior:
1. install latest version of catfish nightly
2. install libconnman-qt5 if it isn't already
3. deploy analog-weather-glow watchface
4. observe that no watchface shows up

**Expected behavior**
What used to happen was that the watchface would display properly.
 
**Device (please complete the following information):**
 - `catfish` but I believe this affects all watches with WiFi
 - AsteroidOS Builddate 11 March 2023 (built on my machine just now)
 - Version: 1.1 nightly

**Additional context**
The watch had successfully been using the `011-analog-silver-swerver` watchface.  I then used `watch-util` to change to the `analog-weather-glow` watchface and this is an extract of the log:

```
Apr 28 15:20:28 catfish systemd[1]: Started OpenSSH Per-Connection Daemon (192.168.2.1:59156).
Apr 28 15:20:28 catfish sshd[4101]: pam_unix(sshd:auth): user [ceres] has blank password; authenticated without it
Apr 28 15:20:28 catfish sshd[4101]: Accepted none for ceres from 192.168.2.1 port 59156 ssh2
Apr 28 15:20:28 catfish sshd[4101]: pam_unix(sshd:session): session opened for user ceres(uid=1000) by (uid=0)
Apr 28 15:20:28 catfish systemd-logind[454]: New session c58 of user ceres.
Apr 28 15:20:28 catfish systemd[1]: Started Session c58 of User ceres.
Apr 28 15:20:28 catfish sshd[4104]: Received disconnect from 192.168.2.1 port 59156:11: disconnected by user
Apr 28 15:20:28 catfish sshd[4104]: Disconnected from user ceres 192.168.2.1 port 59156
Apr 28 15:20:28 catfish sshd[4101]: pam_unix(sshd:session): session closed for user ceres
Apr 28 15:20:28 catfish systemd[1]: sshd@57-192.168.2.15:22-192.168.2.1:59156.service: Deactivated successfully.
Apr 28 15:20:28 catfish systemd[1]: session-c58.scope: Deactivated successfully.
Apr 28 15:20:29 catfish systemd-logind[454]: Removed session c58.
Apr 28 15:20:29 catfish dbus-daemon[444]: [system] Rejected send message, 3 matched rules; type="method_call", sender=":1.567" (uid=1000 pid=4035 comm="/usr/bin/asteroid-launcher -plugin evdevtouch:/dev" label="kernel") interface="org.freedesktop.DBus.Introspectable" member="Introspect" error name="(unset)" requested_reply="0" destination="net.connman" (uid=0 pid=511 comm="/usr/sbin/connmand -n" label="kernel")
Apr 28 15:20:29 catfish asteroid-launcher[4035]: QDBusPendingReply: type ConnmanObjectList is not registered with QtDBus
Apr 28 15:20:29 catfish kernel: binder: undelivered transaction 35141, process died.
Apr 28 15:20:29 catfish kernel: binder: undelivered transaction 35140, process died.
Apr 28 15:20:29 catfish systemd[3410]: asteroid-launcher.service: Main process exited, code=killed, status=6/ABRT
Apr 28 15:20:29 catfish systemd[3410]: asteroid-launcher.service: Failed with result 'signal'.
Apr 28 15:20:29 catfish sensorfwd[500]: [SocketHandler]: Socket error:  QLocalSocket::PeerClosedError
Apr 28 15:20:29 catfish sensorfwd[500]: [SocketHandler]: Noticed lost session:  82
```

I did a `git bisect` and found that it is commit 50c88f818c8f551773f47f983ce1fe6456dbcc95 of `libconnman-qt5` that introduces the problem.  Also note that somewhere since version `1.2.35` the name of the module changes from `Connman` to `MeeGo.Connman` so for the test above, this change has been made to the watchface QML code.

beroset, 2023-03-11:
I think the relevant messages are these:
```
Apr 28 16:51:17 catfish dbus-daemon[444]: [system] Rejected send message, 3 matched rules; type="method_call", sender=":1.971" (uid=1000 pid=6425 comm="/usr/bin/asteroid-launcher -plugin evdevtouch:/dev" label="kernel") interface="org.freedesktop.DBus.Introspectable" member="Introspect" error name="(unset)" requested_reply="0" destination="net.connman" (uid=0 pid=511 comm="/usr/sbin/connmand -n" label="kernel")
Apr 28 16:51:17 catfish asteroid-launcher[6425]: QDBusPendingReply: type ConnmanObjectList is not registered with QtDBus
```
But it appears to me that `ConnmanObjectList` is [actually registered](https://github.com/sailfishos/libconnman-qt/blob/master/libconnman-qt/commondbustypes.h#L38-L48) so it's not clear why this error message occurs.

beroset, 2023-03-11:
If I comment out [the `connect` line](https://github.com/sailfishos/libconnman-qt/blob/master/libconnman-qt/networktechnology.cpp#L100-L115) in `NetworkTechnology::initialize()`, it no longer crashes.  Obviously it also doesn't work, but it appears to be progress.  The log file now contains this:

```
Mar 11 15:22:30 catfish dbus-daemon[444]: [system] Rejected send message, 3 matched rules; type="method_call", sender=":1.1476" (uid=1000 pid=9430 comm="/usr/bin/asteroid-launcher -plugin evdevtouch:/dev" label="kernel") interface="org.freedesktop.DBus.Introspectable" member="Introspect" error name="(unset)" requested_reply="0" destination="net.connman" (uid=0 pid=511 comm="/usr/sbin/connmand -n" label="kernel")
Mar 11 15:22:30 catfish systemd-logind[454]: Removed session c132.
Mar 11 15:22:33 catfish mce[425]: modules/display.c: mdy_blanking_off_cb(): display state req: LPM_ON
Mar 11 15:22:33 catfish mce[425]: modules/display.c: mdy_display_state_leave(): current display state = POWER_DOWN
Mar 11 15:22:33 catfish asteroid-launcher[9430]: sleepDisplay
```
So although there is still the `Rejected send message` error, there is no longer a `ConnmanObjectList` error.

MagneFire, 2023-03-27:
Fixes have been implemented via https://github.com/AsteroidOS/meta-asteroid/pull/138.
Great work @beroset!

closed 2023-03-27 by MagneFire