AsteroidOS/asteroid #254: [hoki] Issue: Watchface does not update time/date correctly when entering Standby Mode

daveflorek1, 2023-06-17:
**Describe the bug**
The watchface does not update the time and date appropriately while in standby mode. When I exit standby mode (button press or screen tap), then watchface will then update the time and date with the stored information.

**To reproduce**
Steps to reproduce the behavior:
1. Update the time and date to preferred settings
2. Wait for the watch to go into Standby Mode where only the time and date are displayed with no background wallpaper
3. Observe the time and date on the watch face drift out of sync until you press the button to exit Standby Mode

**Expected behavior**
I should be seeing the time (and date) automatically updating.

**Screenshots**
(I need guidance on this)

**Device (please complete the following information):**
 - Watch Codename: hoki
 - AsteroidOS Builddate: 20230616041212
 - Version 1.1-nightly

MagneFire, 2023-06-17:
Thank you for reporting this issue.

Unfortunately this is a known issue: https://github.com/AsteroidOS/meta-smartwatch/issues/184
This is also the reason why 'Always-on-Display' is marked as 'partial' on the watch installation page: https://asteroidos.org/watches/hoki/

MagneFire, 2024-11-10:
This has been fixed as of https://github.com/AsteroidOS/meta-smartwatch/pull/247