AsteroidOS/asteroid #257: [Feature Request] Brightness or even adaptive brightness setting for Always On Display

Ragebone, 2023-06-17:
**Is your feature request related to a problem? Please describe.**
My Beluga seems to be very bright when displaying the time on the AOD. 
That is disturbing at night when i want to sleep. So i can't use AOD with Beluga. 
Sturgen on the other hand might be dim enough by default to not be distracting. 
But i find the settings and in this case the settings for the AOD to be lacking. 

**Describe the feature you'd like**
I would like to see an option in the Settings->Display menu to set the brightness for the AOD or to enable Automatic Brightness functionality for the AOD.

beroset, 2024-05-14:
The current Settings->Display menu has:

- Brightness (%)
- Automatic Brightness (checkbox)
- Always On Display (checkbox)
- Burn-in Protection (checkbox)
- Tilt-to-wake (checkbox)
- Tap-to-wake (checkbox)

Setting the brightness can be done simply by adjusting it (Brightness setting) or by enabling Automatic Brightness.  Since these already exist and apply for both AOD and regular mode, it's not clear that anything is really missing.

Ragebone, 2025-01-24:
@beroset 

I am very well aware that there is a setting for Brightness already.

What is not there is one such setting for the AOD, the "Always On Display". 
The AODs brightness is not affected by that existing Brightness setting. 
Beluga sadly defaults to eye scorching maximum brightness on the AOD, that is why i am asking for this

beroset, 2025-01-24:
I think we might be having a terminology problem.  

With **Always-on-Display** (AOD) *disabled*, the watch will blank the display after a few seconds to save battery.  With AOD *enabled*, this behavior is changed and the display does not turn entirely off.  Internally, that transition is signaled by a boolean variable named `displayAmbient`.  That variable is `false` when the display is displaying normally or `true` when the watch would have blanked the display.   However, each individual watchface may make an adjustment at the point the watch would have turned off.  As an example of this, consider the [analog-50s-americana](https://github.com/AsteroidOS/asteroid-launcher/blob/42fe48162daaf8be5ca24bdd6c859770635dc3ae/src/watchfaces-preview/160/014-analog-50s-americana.png) watchface.  Normally, the numbers and hands are black, but they change to white when `displayAmbient` is true.  Further, looking at [this line of watchface code](https://github.com/AsteroidOS/asteroid-launcher/blob/42fe48162daaf8be5ca24bdd6c859770635dc3ae/src/watchfaces/014-analog-50s-americana.qml#L153) we can see that the fill color not only changes from black to white, but also the `alpha` (essentially, the brightness) from 1 (100%) to 0.7 (70%).  

The Display Brightness setting affects both in that it represents the 100% which the watchface itself (they're all different) may further attenuate.

The second feature is **Nightstand** mode which is more or less completely separate.  In Settings->Nighstand we can see that there is a checkbox to enable the feature.  If it is enabled, the Brightness, Delay and Always on Display settings can be adjusted there *just for Nightstand mode*.  

To see how this actually works in practice, try this:

1. Make sure the watch is _not_ on its charger
2. In Settings->Watchface, set the watchface to analog-50s-americana
3. In Settings->Display, set the brightness for 100% and uncheck (disable) Always on Display
4. In Settings->Nightstand, Enable the feature, set Brightness to 10%, set Delay to 5s, enable Always on Display and disable Custom watchface
5. Go back to the watchface display and verify that the brightness is 100%
6. Wait about 12 seconds and verify that the display turns off
7. Put the watch on the charger
8. Verify that, possibly after a few seconds, the watch goes into Nightstand mode and a ring of green charger indicators appears on the outside edge of the watchface
9. Verify that the brightness is 10%
10. Wait about 12 seconds and verify that the display is still on

So with that explanation, I wonder if what you're calling AOD is actually Nighstand mode.  If I have misunderstood you, perhaps you could create a series of steps like the ones above that would allow me to better understand what you're seeing and how you think it should work.

DexasElectronics, 2025-02-09:
Hello. Problem is, that when watch goes into "always-on-display" mode, which is "dim", its still too bright and there isnt a setting to adjust the brightness in the "dim" mode. I assume its set to someting like 10% (somewhere in the firmware), but it would be nice to have a setting to adjust it to whatever value you want.

aspauldingcode, 2025-04-30:
I like this idea to prevent burn-in.