AsteroidOS/asteroid #261: [lenok] Watch wakes up on palm detect sensor

z3ntu, 2023-07-20:
**Describe the bug**
On lenok the "palm sensor" (just putting the whole palm on the display) makes the watch wake up, even if Settings->Display "Tap-to-wake" (and "Tilt-to-wake") is disabled. I guess that should be disabled also?

**To reproduce**
Steps to reproduce the behavior:
1. Disable Tap-to-wake and Tilt-to-wake setting
2. Place palm on watch face
3. Observe watch waking up

**Expected behavior**
The watch shouldn't wake up if Tap-to-wake is disabled.

**Device (please complete the following information):**
 - Watch Codename: lenok
 - AsteroidOS Builddate 2023-07-14?
 - Version 1.1-nightly 20230714031406

MagneFire, 2023-07-20:
Thanks for the bug report!

Can you confirm when this happens?

If you're using Always-on-Display then this is expected for about one second after entering Always-on-Display mode. That's because we give the compositor some time to fully render the display before suspending display redraws.

z3ntu, 2023-07-21:
This always seems to happen, also when Always-on-Display is turned off (as on @PureTryOut's watch). It also happens long after the screen has been off already (or in always-on mode).

We've also noticed now that using the palm also makes the screen turn off when it's on, so it goes both ways.