AsteroidOS/asteroid #263: [beluga] System crashes immediately after boot

nerdius, 2023-07-30:
I managed to brick my OPPO Watch a week ago after I flashed a new firmware I downloaded at that time. I tried using `fastboot` and TWRP to wipe the partitions and reinstall the images, re-flashed it several times, cleaned contacts, used different ports on my computer, but to no avail. It behaves exactly the same way.

When I boot the device, the boot splash screen appears. For a brief moment, I can access ADB shell, then screen goes black and I get disconnected.

I ran the following command to repeatedly get `dmesg` output and dump it to a file:

```
while adb -s SERIAL_NO shell dmesg >> beluga-dmesg.txt; do
  echo -e "\n- - - - -\n" >> beluga-dmesg.txt
  sleep 0.5
done
```

This is what I got: _[beluga-dmesg.txt](https://github.com/AsteroidOS/asteroid/files/12210213/beluga-dmesg.txt)_

I'm still trying to figure out what's going on...

wannaphong, 2023-07-31:
Do you has unlock bootloader the watch in WearOS before flash?

nerdius, 2023-07-31:
@wannaphong,yes, of course. Otherwise, I wouldn't be install it. I had AsteroidOS for a few months, and I decided to re-flash it a week ago to update the firmware because the package manager kept complaining about the dependencies.

wannaphong, 2023-07-31:
Can you join the chat? https://app.element.io/#/room/#asteroid:matrix.org

nerdius, 2023-07-31:
@wannaphong, I'm busy right now, but yes, good idea. I didn't know there was a Matrix channel.

MagneFire, 2023-07-31:
It looks like your `vendor` partition is corrupt. You need to restore that by flashing the stock WearOS images.

What's the model number of your watch? So I can attempt to find these images.

nerdius, 2023-07-31:
@MagneFire, the exact model is _OPPO Watch 46 mm Wi-Fi-only_.

That was my plan too. I was going to flash the original firmware, re-lock the bootloader to check integrity of the firmware, then unlock it and install AsteroidOS normally. Unfortunately, I couldn't find the firmware images online.

MagneFire, 2023-07-31:
@nerdius Thanks for confirming. I have this v51 build for it: https://mega.nz/file/vc8GQSbZ with decryption key `SvgFk7Bt4gbT2yIVfMelDz6vvxikEAy9WU0N1iUgP1A`.

nerdius, 2023-07-31:
@MagneFire, thanks a lot.

I also found [partition backups](https://mega.nz/file/tlpXHbLZ#MUIW9buNL1TSWheH37VszFe215zaTr01QSJPrjlpSp0) on XDA Forum.

nerdius, 2023-07-31:
Re-flashing stock _vendor_ image, and then AsteroidOS _userdata_ and _boot_ didn't fix anything.

How can I restore the stock firmware so I can start from scratch? Is there a command in `fastboot` to flash all partitions at once?

MagneFire, 2023-07-31:
> Re-flashing stock _vendor_ image, and then AsteroidOS _userdata_ and _boot_ didn't fix anything.
> 
> How can I restore the stock firmware so I can start from scratch? Is there a command in `fastboot` to flash all partitions at once?

To start from scratch you can flash the following partitions, these were flashed in the past and confirmed working by multiple people:
```
fastboot flash system system.img 
fastboot flash vendor vendor.img
fastboot flash userdata userdata.img
fastboot flash recovery recovery.img
fastboot flash cache cache.img
fastboot flash boot boot.img
```

nerdius, 2023-07-31:
@MagneFire, I booted to Wear OS and even re-locked bootloader, after which Wear OS reset itself and booted back up normally.

Then I unlocked the bootloader and flashed AsteroidOS's _userdata_ and _boot_ images. It still doesn't boot but this time, it's just stuck at the boot screen; it doesn't go black and I don't get disconnected from ADB shell.

I also tried the temporary install method, and this one didn't work either.

Here's what I get on `dmesg` (last 40 lines):

```
[  274.626692] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.627701] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.627859] msm_pcm_volume_ctl_get substream or runtime not found
[  274.629176] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.629483] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.634290] msm_pcm_volume_ctl_get substream not found
[  274.634670] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.634999] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.635317] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.636550] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.636892] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.638090] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.638415] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.639645] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.640000] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.643997] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.644363] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.645508] msm_pcm_volume_ctl_get substream or runtime not found
[  274.646171] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.646507] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.647759] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.648080] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.649348] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.649696] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.653443] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.653823] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.655133] msm_compr_audio_effects_config_get: stream or effects inactive
[  274.655485] msm_adsp_stream_callback_get: ASM Stream PP event queue is not initialized.
[  274.824702] init: Received control message 'interface_start' for 'android.hardware.power@1.2::IPower/default' from pid: 643 (/system/bin/hwservicemanager)
[  274.824747] init: Could not find service hosting interface android.hardware.power@1.2::IPower/default
[  274.843882] init: Received control message 'interface_start' for 'vendor.qti.hardware.perf@2.0::IPerf/default' from pid: 643 (/system/bin/hwservicemanager)
[  274.843927] init: Could not find service hosting interface vendor.qti.hardware.perf@2.0::IPerf/default
[  274.853402] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[  274.853416] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[  274.853427] send_afe_cal_type cal_block not found!!
[  274.868523] Active_profile:hq, next_profile:hq
[  274.869131] Startup of device [left] is a warmstart
[  274.870212] q6asm_send_cal: cal_block is NULL
[  274.880020] init: Received control message 'interface_start' for 'vendor.qti.hardware.perf@2.0::IPerf/default' from pid: 643 (/system/bin/hwservicemanager)
[  274.881298] init: Could not find service hosting interface vendor.qti.hardware.perf@2.0::IPerf/default
```

MagneFire, 2023-07-31:
Thanks for the log. I'll try to reproduce this in the coming days.

For completeness, can you send me the `logcat` (`adb shell "system/bin/logcat"`) and the journal log (`adb shell "journalctl --no-pager"`) as well?

nerdius, 2023-07-31:
@MagneFire, I re-flashed it several times, but it's still the same. All logs complain about _/vendor_ a lot.

Here are all the logs in a single file: _[beluga-logs.txt](https://github.com/AsteroidOS/asteroid/files/12221991/beluga-logs.txt)_

MagneFire, 2023-07-31:
Indeed interesting errors.

Can you add the following to the `/init.rc` file and test again (reboot)?

```
service vendor.power-hal-1-0 /vendor/bin/hw/android.hardware.power@1.0-service
    class hal
```

nerdius, 2023-07-31:
@MagneFire, sure. At the very end, I assume.

MagneFire, 2023-07-31:
Hmm, looking at the logs again, I think it's something else:

```
Apr 28 17:42:42 beluga asteroid-launcher[825]: qt.qpa.plugin: Could not load the Qt platform plugin "hwcomposer" in "" even though it was found.
Apr 28 17:42:42 beluga asteroid-launcher[825]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Apr 28 17:42:42 beluga asteroid-launcher[825]: Available platform plugins are: hwcomposer, minimal, offscreen, vnc, wayland-egl, wayland.
Apr 28 17:42:42 beluga systemd[757]: asteroid-launcher.service: Main process exited, code=killed, status=6/ABRT
Apr 28 17:42:42 beluga systemd[757]: asteroid-launcher.service: Failed with result 'signal'.
```

This would indicate an error with the Qt plugin system.

Which images are you using? We've updated ours yesterday: https://release.asteroidos.org/nightlies/beluga/

nerdius, 2023-07-31:
@MagneFire, after adding the two lines you suggested, I'm getting the exact same behaviour, even in logs.

I'm using images I downloaded at: _2023-07-21 19:23:36_. I didn't know the _nightlies_ directory went back online. I'll try the new images.

nerdius, 2023-07-31:
And I have a welcome screen!

[image]

Not if I needed to flash Wear OS to begin with, but after flashing the new images, it works again.

@MagneFire, thank you and entire AsteroidOS community for the effort! We need to have some plan to commercialize AsteroidOS to bring it to masses. :smile:

closed 2023-08-01 by nerdius

teodorrg, 2024-03-05:
Hello, can someone help me with the backup files for Oppo Watch 46mm (Wi-fi) (OW19W8)? I tried to root it and ended up deleting everything from it... now it's displaying "No OS system...".