AsteroidOS/asteroid #264: [beluga] Bluetooth not discoverable

Unb0rn, 2023-08-04:
**Describe the bug**
On latest nightly (04.08) bluetooth is not discoverable. Tried two different phones with both AsteroidOS Sync and Gadgetbridge.
Switch various options in `bluetoothctl` didn't seem to help. Switching Host Mode from `le` to `dual` solved the issue.

**To reproduce**
Steps to reproduce the behavior:
1. Install the latest nightly
2. Enable bluetooth in settings
3. Try to discover the device on any companion app
4. See nothing)

**Expected behavior**
The device is discoverable in the companion app

**Device (please complete the following information):**
 - Watch Codename: beluga
 - AsteroidOS Builddate: 2023-08-03
 - Version: 1.1 nightly

easyKL, 2023-08-08:
Same here.
Not discoverable by three different devices.
Codename beluga and neither Gadgetbridge nor AsteroidOS Sync could.

Ragebone, 2024-01-07:
Beluga owner here, i have two, and can say that they work for me and i think i am on the 1.1 nightly. 

I had issues pairing them that turned out to be android 13 bug. 
Wrote about it here: https://github.com/AsteroidOS/asteroid/issues/253
I am running IodeOS and they fixed that Bluetooth issue in literally the next update.
So that all has been working for me for at least half a year now.