AsteroidOS/asteroid #265: [Idea] Asteroid Mobile

GreenLunar, 2023-08-08:
A First-Class Easy UI Experience Baked Into Your Handset So Well Your Grandmother Could Use It

You migh have heard of Plasma Mobile (KDE Plasma) and GNOME Mobile (GNOME) and Phosh (GNOME) UIs for tablets and handsets.

**Is your feature request related to a problem? Please describe.**
New edition of Asteroid called Asteroid Mobile for PDAs, Handsets and Remote Controls.

**Describe the feature you'd like**
SMS and Phone programs, but not before adding IRC, Matrix and XMPP clients.

**Why**
The size of the buttons is good, and the UI options are good especially for PDA devices for the elderlies, the disabled and children.

**Additional context**
https://gitlab.com/postmarketOS/pmaports/-/issues/2250

Currently testing Asteroid with Samsung Galaxy SII (I9100) using postmarketOS.

GreenLunar, 2023-08-10:
Watch video https://imgur.com/a/4IbsRDc (50 seconds)

For some reason, this website doesn't allow MKV files.

sorry-i-am-late, 2024-05-16:
I would be a fan assuming my other programs worked well. 

Since it is designed for low spec watches i assume it would speed up my pinephone a good bit in terms of ui snappiness. If you get sms and calls working this could be a huge advancement in terms of smooth mobile linux UIs; an area that doesnt see alot of new competition that looks anywhere near as slick as this.

I wonder how well it would work just on my Pinephone...

GreenLunar, 2024-05-16:
> I would be a fan assuming my other programs worked well.

I do not know yet. SSH does not work yet, because something in the Asteroid interface appears to interrupt it.

Because the Asteroid interface is based on Glacier, then I assume that almost any program that you would need should be installable.

> I wonder how well it would work just on my Pinephone.

I would say, very smooth.

The device I have tested this with is substantially not as powerful as PinePhone.