AsteroidOS/asteroid #267: Fails on Wifi-Enabler

phillippi2, 2023-08-18:
I am trying to build AsteroidOS on Arch Linux. This is for the Asus Zenwatch 3, swift. However, the build fails when running wifi-builder.
The error message state "Error: nothing provides wifi-enabler." Then it says that the dependency is missing.

I downloaded Asteroid using got clone on its main repository. Then I built it. Finally, tried to build it using the suggested command: bitbake asteroid-image.

On a side note, figuring out how to set bitbake up is a bit frustrating because of the lack of documentation online. The one from Arch's repository fails to build. I'm still not sure I got it setup properly, as it seems to depend on the location it's located in. Anyways.

What can I try to alleviate this problem?

MagneFire, 2023-08-18:
When did you clone the repository? The `wifi-enabler` recipe was added two days ago: https://github.com/AsteroidOS/meta-asteroid/commit/cf300247bd3a0913bf4656da5ada4ca5a5910bbc

To easy up on building AsteroidOS you can also build it using Docker or Podman: https://wiki.asteroidos.org/index.php/Building_AsteroidOS#Build_with_containers.

Also, you only need to ensure that the dependencies are installed on your Arch setup. `bitbake` is downloaded as part of the `prepare-build.sh` script. For building without containers you can also follow this guide: https://wiki.asteroidos.org/index.php/Building_AsteroidOS#Build_without_containers

phillippi2, 2023-08-18:
> When did you clone the repository? The `wifi-enabler` recipe was added two days ago: [AsteroidOS/meta-asteroid@cf30024](https://github.com/AsteroidOS/meta-asteroid/commit/cf300247bd3a0913bf4656da5ada4ca5a5910bbc)
> 
> To easy up on building AsteroidOS you can also build it using Docker or Podman: https://wiki.asteroidos.org/index.php/Building_AsteroidOS#Build_with_containers.
> 
> Also, you only need to ensure that the dependencies are installed on your Arch setup. `bitbake` is downloaded as part of the `prepare-build.sh` script. For building without containers you can also follow this guide: https://wiki.asteroidos.org/index.php/Building_AsteroidOS#Build_without_containers

I had cloned the repository on Sunday. I'll try the new version when I get back home. I'll update after trying it.

phillippi2, 2023-08-19:
Update: I recloned the repository and tried to compile it. It shows that it prepared bitbake. However, the system acts like it didn't install it. Cannot find the command. I'll see if *

phillippi2, 2023-08-19:
I finally got to try to compile AsteroidOS. Now, I get another error. However, I don't understand what the error is. Let's see if I can figure out how to post code manageably.

     ~/asteroid    master  bash -c "source ./prepare-build.sh dory && bitbake asteroid-image"         126 ✘ 
Writing build/conf/local.conf
Writing build/conf/bblayers.conf
Welcome to the Asteroid compilation script.

If you meet any issue you can report it to the project's github page:
    https://github.com/AsteroidOS

You can now run the following command to get started with the compilation:
    bitbake asteroid-image

Have fun!
Loading cache: 100% |                                                                               | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##############################################################################| Time: 0:00:23
Parsing of 3036 .bb files complete (0 cached, 3036 parsed). 4496 targets, 463 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "manjaro"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "dory"
DISTRO               = "asteroid"
DISTRO_VERSION       = "1.1-nightly"
TUNE_FEATURES        = "arm armv7ve vfp neon callconvention-hard"
TARGET_FPU           = "hard"
meta-wren            
meta-triggerfish     
meta-tetra           
meta-swift           
meta-sturgeon        
meta-sprat           
meta-sparrow         
meta-smelt           
meta-skipjack        
meta-sawfish         
meta-ray             
meta-pike            
meta-nemo            
meta-narwhal         
meta-mtk6580         
meta-mooneye         
meta-minnow          
meta-lenok           
meta-koi             
meta-hoki            
meta-dory            
meta-catfish         
meta-beluga          
meta-bass            
meta-anthias         = "master:8a13d809b0bb22c0e9fbefbaf19ff8f478315c8d"
meta-filesystems     
meta-python          = "HEAD:36707815060e387c2370962b5eb21bac7750f924"
meta-android         = "HEAD:038458fd1fe3d946ce084865adb09238c12eaf90"
meta-networking      
meta-gnome           
meta-multimedia      
meta-oe              = "HEAD:36707815060e387c2370962b5eb21bac7750f924"
meta-asteroid-community = "master:1c6adf7e85ad1b3281770c7cebf3b19cbb66a6d6"
meta-asteroid        = "master:cf300247bd3a0913bf4656da5ada4ca5a5910bbc"
meta                 = "HEAD:4766ba017b7562e42b33fde7f2e84c2a339e3f4c"
meta-qt5             = "HEAD:bff5bd937f0776166e81a63f3dd39ede348ef758"

Initialising tasks: 100% |###########################################################################| Time: 0:00:02
Sstate summary: Wanted 2479 Local 0 Mirrors 0 Missed 2479 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
WARNING: source-han-sans-cn-fonts-1.004-r0 do_fetch: Failed to fetch URL https://github.com/adobe-fonts/source-han-sans/raw/release/SubsetOTF/SourceHanSansCN.zip, attempting MIRRORS if available
WARNING: source-han-sans-kr-fonts-1.004-r0 do_fetch: Failed to fetch URL https://github.com/adobe-fonts/source-han-sans/raw/release/SubsetOTF/SourceHanSansKR.zip, attempting MIRRORS if available
ERROR: iso-codes-4.13.0-r0 do_package_write_ipk: Fatal errors occurred in subprocesses:
Command 'PATH="/home/username/asteroid/src/oe-core/scripts:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/bin/allarch-oe-linux:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot/usr/bin/crossscripts:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/sbin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/bin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/sbin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/bin:/home/username/asteroid/src/oe-core/bitbake/bin:/home/username/asteroid/build/tmp-glibc/hosttools" opkg-build -Z xz -a "--memlimit=50% --threads=16" iso-codes-locale-zh-tw /home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/deploy-ipks/all' returned non-zero exit status 1.
Subprocess output:find: paths must precede expression: `SHLVL=0'
Command 'PATH="/home/username/asteroid/src/oe-core/scripts:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/bin/allarch-oe-linux:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot/usr/bin/crossscripts:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/sbin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/usr/bin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/sbin:/home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/recipe-sysroot-native/bin:/home/username/asteroid/src/oe-core/bitbake/bin:/home/username/asteroid/build/tmp-glibc/hosttools" opkg-build -Z xz -a "--memlimit=50% --threads=16" iso-codes-locale-zh-hk /home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/deploy-ipks/all' returned non-zero exit status 1.
[... cut ...]
ERROR: Logfile of failure stored in: /home/username/asteroid/build/tmp-glibc/work/all-oe-linux/iso-codes/4.13.0-r0/temp/log.do_package_write_ipk.1338710
ERROR: Task (/home/username/asteroid/src/oe-core/meta/recipes-support/iso-codes/iso-codes_4.13.0.bb:do_package_write_ipk) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3186 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/username/asteroid/src/oe-core/meta/recipes-support/iso-codes/iso-codes_4.13.0.bb:do_package_write_ipk
Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

argosphil, 2023-08-22:
Maybe you could try reproducing the problem on an arch live system booted from an .iso, if such a thing exists? We tried doing that for @eLtMosen 's distro but it worked fine using the .iso...

I suspect it's a toolchain issue similar to the perl one. (My understanding is perl removed a widely-used feature still marked as experimental, and bitbake assumed perl was stable enough to use host perl rather than building a perl-native recipe).

argosphil, 2023-08-23:
I've managed to reproduce this by installing EndeavourOS in `qemu`. It happens with the system-supplied bash, which is bash version 5.1.16(1)-release, built with GCC 13.1.1 20230429, arch release 4. Rebuilding the package with `makepkg` and GCC 13.2.1 20230801 makes the problem go away, even if bash 5.1.016-4 is built.

I'm not convinced this is a bash or GCC bug yet. It might also be a libpseudo interaction or another library bug triggered by a specific combination of factors.