AsteroidOS/asteroid #272: [qemux86 et. al.] Build failed due to incomplete documentation and libxcrypt

andidevi, 2023-09-24:
**Describe the bug**

I tried to build AsteroidOS following the instructions [here](https://wiki.asteroidos.org/index.php/Building_AsteroidOS) but replacing dory with qemux86.

**To reproduce**
Steps to reproduce the behavior:
1. Follow the build instructions on https://wiki.asteroidos.org/index.php/Building_AsteroidOS on arch linux
2. Watch it fail on libxcrypt because of missing **python-passlib**
3. install python-passlib, ie. do `yay -Sy  python-passlib`
5. clear all build caches (may be optional)
6. see the build fail again on libxcrypt. this time it's about wrong command line parameters:

**Expected behavior**
1. Build instructions on https://wiki.asteroidos.org/index.php/Building_AsteroidOS should mention python-passlib in command line for Arch linux under _Distribution prerequisite installation_ 
2. Build succeeds, ie. libxcrypt uses correct command line for python 3.11 and python-passlib 1.7

**Screenshots**
see attached logs

**Device (please complete the following information):**
 - Watch Codename: qemux86 but other probably affected too, as this is a host tools error
 - AsteroidOS Builddate: 2023-09-24
 - Version: 1.1 nightly 

**Additional context**
[bitbake-cookerdaemon.log](https://github.com/AsteroidOS/asteroid/files/12708485/bitbake-cookerdaemon.log)
[config.log of libxcrypt build](https://github.com/AsteroidOS/asteroid/files/12708487/config.log)
[bitbake-console-output.log](https://github.com/AsteroidOS/asteroid/files/12708491/bitbake-console-output.log)

Charlyo, 2023-11-15:
@andidevi if you haven't found the solution yet, it's a version compatibility between libxcrypt and pearl. 

My workaround was to ensure the most current commit of libxcrypt is fetched by editing: `asteroid/src/oe-core/meta/recipes-core/libxcrypt/libxcrypt.inc`

Inside, change `SRCREV` to `SRCREV = "${AUTOREV}"`

MagneFire, 2023-12-25:
Thanks for reporting, I was able to reproduce this on Arch Linux as well.

A fix has been included in https://github.com/AsteroidOS/meta-asteroid/pull/181. Feel free to reopen if you still encounter this issue.

closed 2023-12-25 by MagneFire