AsteroidOS/asteroid #274: [catfish] Fail on real installation

SebSK3, 2023-10-28:
When trying to real install it fails.

**To reproduce**
Steps to reproduce the behavior:
1. Perform everything like described on here until point 2.2 https://asteroidos.org/watches/catfish/
2. On command: `fastboot flash userdata ~/Downloads/asteroid-image-catfish.ext4`
It fails with this output:
```
[sebsk@nowhere asteroid-hosttools]$ fastboot flash userdata ~/Downloads/asteroid-image-catfish.ext4
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::compare: __pos (which is 464) > this->size() (which is 0)
zsh: IOT instruction (core dumped)  fastboot flash userdata ~/Downloads/asteroid-image-catfish.ext4
```

**Expected behavior**
It should install

**Device (please complete the following information):**
 - Watch Codename: catfish (2018 version)
 - AsteroidOS Builddate [e.g. 2022-03-06 (leave empty if you don't know)]:
 - Version [e.g. 1.1 nightly / stable]: stable

**Additional context**
Temporary installation works without issues. Things i tried:
- Removing asteroidos.ext4 file after temporary install
- Factory resetting watch
- Using asteroid-hosttools for install (command: `flashy catfish --nightly` ) (same error)

MagneFire, 2023-10-29:
Are you attempting this on an Arch Linux based system?
This appears to be a recent issue with `fastboot`. See: https://bugs.archlinux.org/task/79685?project=1&openedfrom=-1+week.
It's currently recommended to downgrade the `android-tools` package to an older version.

SebSK3, 2023-10-29:
> Are you attempting this on an Arch Linux based system? This appears to be a recent issue with `fastboot`. See: https://bugs.archlinux.org/task/79685?project=1&openedfrom=-1+week. It's currently recommended to downgrade the `android-tools` package to an older version.

Thanks for guiding me in good direction! I am indeed using Arch Linux. Downgrading `android-tools` made this error disappear, but new one to appear, so for people with this problem, here are steps i made to succesful flash:

1. Download platform tools from here (for your platform): https://developer.android.com/tools/releases/platform-tools
2. Unzip
3. Inside folder, in terminal, type `./adb devices` so deamon will start
4. Follow instructions on asteroidos install, but use downloaded fastboot

closed 2023-10-29 by SebSK3