AsteroidOS/asteroid #278: [hoki] Flashing userdata stuck at sending sparse file

rhetenor, 2024-01-17:
**Describe the bug**
When trying to flash the userdata of my Fossil Gen6 with the ext4 image under Arch Linux I first encountered the error:

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::compare: __pos (which is 464) > this->size() (which is 0)
```

but after some searching it turns out there is a bug in the distributed fastboot from the  `android-tools` package under Arch. So I downloaded the native `platform-tools` from Google directly which seem to work. (Just wanted to note this if anybody else encounters the error)

Unfortunately the process now gets stuck when sending the sparse files. When I flash the image files with: 
`./fastboot flash userdata ~/Downloads/asteroid-image-hoki.ext4`

It gets stuck at:

`Sending sparse 'userdata' 1/4 (251449 KB) `

After 30 Minutes I've interrupted the process.

Reducing the sparse file size to 1M for testing with the `-S` option of fastboot leads to the following output

```
Sending sparse 'userdata' 1/783 (1020 KB)          OKAY [  0.037s]
Writing 'userdata'                                 OKAY [  0.153s]
Sending sparse 'userdata' 2/783 (1020 KB)          OKAY [  0.034s]
Writing 'userdata'                                 OKAY [  0.027s]
Sending sparse 'userdata' 3/783 (1020 KB)          OKAY [  0.034s]
Writing 'userdata'                                 OKAY [  0.050s]
Sending sparse 'userdata' 4/783 (1020 KB)          OKAY [  0.034s]
Writing 'userdata'                                 OKAY [  0.026s]
Sending sparse 'userdata' 5/783 (1020 KB)          OKAY [  0.034s]
Writing 'userdata'                                 OKAY [  9.745s]
Sending sparse 'userdata' 6/783 (1020 KB)          OKAY [  0.036s]
Writing 'userdata'                                 OKAY [  0.027s]
Sending sparse 'userdata' 7/783 (1020 KB)          
```

It get's stuck at step 7 (Interrupted after 5 minutes). Increasing the size to 10/25/50M leads it to get stuck at step 5/3/2 respectively.
Also notice the write lag on step 5, which always happens shortly before the sending does not work anymore.

After manually interrupting the process I can't communicate with the device over fastboot anymore and I have to reboot the bootloader.

I don't know if this is a problem with the image, the watch or fastboot but if you have any ideas it would be really appreciated. :)

**Device (please complete the following information):**
 - Watch Codename: hoki
 - Version: latest nightly

**Additional context**
The watch did only run AsteroidOS before on temporary install which worked fine.

rhetenor, 2024-01-18:
Somehow after letting the battery of the watch drain and powering it again the flashing ran through. I have no idea what might cause such behavior but now it works! 

Thanks for your work on AsteroidOS!

closed 2024-01-18 by rhetenor

okias, 2024-02-21:
I had same issue, what helped was (not sure if all steps are needed):
 1. boot Android, wipe to factory
 2. power off the watch
 3. `fastboot -S 200M flash userdata ~/Downloads/asteroid-image-hoki.ext4`