AsteroidOS/asteroid #279: [sturgeon] BLE client mode doesn't discover services on some devices

A-Emile, 2024-01-19:
**Describe the bug**
I am building an app to connect and lock/unlock my escooter via BLE. The Problem is, that after connecting to the scooter the services arnt getting discovered. I have tested other ble devices and the services are getting discovered succefully. nrf connect on my phone is discovering the services correctly.
I am using the simpleble library in cpp.
After connecting the size of services is 0 and mtu size is somehow also 0.
in bluetoothctl the UART services is being listed with the info command but in the gatt menu the list-attributes command also shows nothing.

Any Ideas how to fix that would be greatly appreciated

**To reproduce**
Steps to reproduce the behavior:
1. `get ssh shell`
2. `bluetoothctl`
3. `scan on`
4. `connect <mac address>`
5. `info` (it shows the uart service (sometimes))
6. `menu gatt`
7. `list-attributes` (returns nothing)

**Expected behavior**
`list-attributes` should show me the services and characteristics of the connected device.

**Device (please complete the following information):**
 - Watch Codename: sturgeon
 - Version 1.1-nightly

**Additional context**
btmon log: https://gist.github.com/A-Emile/7d09cea00146529ec96f9d175d55c57a#file-btmon-txt

beroset, 2024-01-26:
To help us work on the problem, I have two requests for more information:

1. please post the contents of `/etc/os-release` from the watch
2. if you temporarily change the line in `/etc/bluetooth/main.conf` from `ControllerMode = le` to `ControllerMode = dual` and reboot the watch, does it seem to work then?

A-Emile, 2024-01-29:
Thanks for the respone!

So here is the conetent of `/etc/os-release`:
```
ID=asteroid
NAME="AsteroidOS"
VERSION="1.1-nightly"
VERSION_ID=1.1-nightly
VERSION_CODENAME="${DISTRO_CODENAME}"
PRETTY_NAME="AsteroidOS 1.1-nightly"
BUILD_ID="20240112002713"
LOGO=logo-asteroidos
HOME_URL="https://asteroidos.org/"
```

And i forgot to mention that i already changed `ControllerMode` to `dual` and it didnt seem change anything..