AsteroidOS/asteroid #285: update from mickledore to scarthgap

beroset, 2024-04-28:
As described in #245, we updated to mickledore last year, but that is not EOL (as of December 2023).  The new long term support version is scarthgap which is expected to be released on 30 April 2024 and is a long term support version with EOL April 2028.

MagneFire, 2024-04-28:
For what it's worth I've done some work upgrading to `nanbield`:
- https://github.com/MagneFire/meta-asteroid/tree/f/nanbield
- https://github.com/MagneFire/asteroid/tree/f/nanbield

In summary, a lot is broken:
- dbus seems non-functional
- systemd is failing to launch services
- opening the `/dev/binder` fails.

I've downgraded systemd to the one we currently use with `mickledore`. Which fixes dbus related issues, but opening the binder interface still fails resulting in the launcher failing to show anything.

Maybe some of these issues are relevant when upgrading to `scarthgap`.

casept, 2024-06-18:
I've done a bit more work and most watches at least build now. I've tested catfish and it boots, but also has the binder issues and `asteroid-launcher` does not start.

MagneFire, 2024-07-01:
I've finally made it all work. The issue seems to originate with `libhybris` and compatibility issues with 64bit time_t support (introduced in https://github.com/openembedded/openembedded-core/commit/b9e0c5e750c3097e176fdc18b3b58b622f716e71).

For now I've added a workaround by adding the following to https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-core/libhybris/libhybris_git.bbappend:
```
TARGET_CC_ARCH:remove = "-D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_TIME_BITS=64"
INSANE_SKIP:append:pn-${PN} = "32bit-time"
```
See https://github.com/MagneFire/meta-asteroid/commit/db993a49d25cd23162f5cb8f66f0a9436c6a6fdf

I'm not entirely sure why this is required. My current best guess is that it's some kind of incompatibility with Android Bionic `time_t` and libc `time_t`. We should probably investigate this further.

beroset, 2024-07-04:
Trying to build this now for catfish.  Thanks for the awesome effort!

beroset, 2024-07-05:
I had two three small problems, all of which I was able to fix.  After fixing these problems, I flashed the image to my `catfish` and it seems to be running just fine.

The problems were:
1. Trouble [building the SDK](https://github.com/AsteroidOS/asteroid/issues/287) due to `cracklib` issues
2. problem retrieving `desktop-file-utils` (apparently they changed from http to https)
3. missing/changed symlink:  instead of `tmp-glibc/deploy/catfish/asteroid-image-catfish.ext4` it was `tmp-glibc/deploy/catfish/asteroid-image-catfish.rootfs.ext4`

The attached file [scarth.txt](https://github.com/user-attachments/files/16110119/scarth.txt) can be applied to the `scarthgap` version of `oe-core` by going to the `oe-core` directory and doing `git apply scarth.txt`.

MagneFire, 2024-08-02:
Finally figured out the issue where `systemd` services would fail on older watches (Linux kernel 3.10).
Adding debug traces to `systemd` revealed that processes seem to crash upon calling `umount2()`.

Performing a `git blame` revealed a logic change in systemd: https://github.com/systemd/systemd-stable/commit/ea0f3289a288affd5f13c83849b984c8fad63e90

Reverting the logic in `mount_switch_root_pivot()` seems to workaround this issue. Interestingly the previous code would fail:
```
Failed to pivot root to new rootfs '/run/systemd/mount-rootfs': Device or resource busy
```
I've tested this old implementation on `catfish` (Linux kernel 3.18) as well where it results in the same error. However in case of failure `mount_switch_root_move()` is called which works in both (3.10 and 3.18) cases.
With the new implementation this error does not occur instead resulting in successfully pivoting root on 3.18 while on 3.10 the process crashes at `umount2`.

I've traced the git tree for differences between the two kernel versions specifically for the `umoun2` and namespace related syscalls. Finally the commit https://android.googlesource.com/kernel/msm/+/8033426e6bdb2690d302872ac1e1fadaec1a5581%5E%21/ seems to solve this issue for at least `sparrow`. I'll test other watches in the coming days.

MagneFire, 2024-08-06:
Alright, I've tested the current work on all my watches.

The following worked fine: `sturgeon`, `sparrow`, `swift`, `smelt`, `hoki`, `beluga`, `narwhal`, `dory`, `tetra`, `sawfish`, `pike`, `ray`, `lenok`
While `minnow` still suffers from a non-working USB and stays on the bootlogo.

MagneFire, 2024-08-06:
See https://github.com/AsteroidOS/asteroid/pull/292 for the main PR and references to the other relevant PRs.

beroset, 2024-08-09:
Merged!  Thanks to all for your efforts.

closed 2024-08-09 by beroset