AsteroidOS/asteroid #287: unable to build SDK on Fedora 40

beroset, 2024-05-25:
**Describe the bug**
I have recently updated my computers from Fedora 39 to Fedora 40 and have found that I can no longer build the SDK.  

**To reproduce**
Steps to reproduce the behavior:
1. Follow the steps to build the SDK as described [here](https://wiki.asteroidos.org/index.php/Building_the_SDK)
2. Execute `bitbake -k meta-toolchain-qt5`  (the `-k` to keep going after the first error)
3. Observe that `libffi`, `pseudo` and `cracklib` fail to build

**Expected behavior**
I expected that the SDK would be built in its entirety without errors.

**Detailed error message**
The tail end of the build attempt as described above looks like this:
```
Summary: 3 tasks failed:
  virtual:native:/home/beroset/tools/AsteroidOS/asteroid/src/oe-core/meta/recipes-support/libffi/libffi_3.4.4.bb:do_compile
  virtual:native:/home/beroset/tools/AsteroidOS/asteroid/src/oe-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_compile
  virtual:native:/home/beroset/tools/AsteroidOS/asteroid/src/oe-core/meta/recipes-extended/cracklib/cracklib_2.9.11.bb:do_compile
Summary: There were 6 ERROR messages, returning a non-zero exit code.
```
**Version:**
This uses the latest version, as of 25 May 2024, which is commit `53ad9fe16548492895d9b50239cb1f7ebffd4d38` of https://github.com/AsteroidOS/asteroid.git

**Additional context**
I don't believe this problem occurred with Fedora 39, but I'm not sure how long it's been since I deleted the entire `build` directory and started from scratch to build the toolchain.  One change is that the (native) version of gcc is now `gcc (GCC) 14.1.1 20240507 (Red Hat 14.1.1-1)` which might be more picky about compiling.   Another change is that wget is now `GNU Wget2 2.1.0` and no longer supports the `--passive-ftp` flag.

MagneFire, 2024-06-02:
For what it's worth this is likely to be fixed when we move to `scarthgap` (https://github.com/AsteroidOS/asteroid/issues/285).

In my case moving to that and manually applying https://github.com/openembedded/openembedded-core/commit/914128f6bd988cde278e087fb9457a0c70c7e5ec solved build issues. But I'm testing on an updated Arch (also with GCC 14).

beroset, 2024-07-05:
I wrote an upstream patch which is now applied to cracklib.  (See [this issue](https://github.com/cracklib/cracklib/issues/74).)  The next step is to submit a patch to yocto.

MagneFire, 2024-08-11:
Is this still an issue or has this been resolved by the scarthgap update?

beroset, 2024-08-12:
Just retested both the emulator and watch SDK build, and both now work.

closed 2024-08-12 by beroset