AsteroidOS/asteroid #298: Update from scarthgap to styhead

MagneFire, 2024-12-15:
Yocto Styhead has been released in October, we should consider updating to this version.

See:
- https://wiki.yoctoproject.org/wiki/Releases
- https://docs.yoctoproject.org/dev/migration-guides/release-5.1.html

MagneFire, 2025-01-05:
Current state on `beluga` is that the kernel seems to work fine, the rootfs has at least one issue: Touch works fine, but interacting with an app causes it to close or crash.

Also meta-openembedded needed a patch in order to actually package. A PR is open here: https://github.com/openembedded/meta-openembedded/pull/915

MagneFire, 2025-01-20:
Currently our recipes and the ones provided by others (mainly `meta-qt5`) cause build failures if we keep failing on `buildpaths` (default in `styhead`, only warning in `scarthgap`).

This results in `in package x contains reference to TMPDIR [buildpaths]` in the build log.

To workaround it I've reverted this back to warning in the `asteroid.conf`:
```
ERROR_QA:remove = "buildpaths"
WARN_QA:append = " buildpaths"
```

It will take quite a lot of effort to get all recipes to not include references to the TMPDIR. Hence I'd like to propose actually applying said change in `asteroid.conf`. Though this does risk us not actually fixing these warnings, as we haven't done so previously either.

A similar thing is currently applied for lack of upstream status in patch files:
```
ERROR_QA:remove = "patch-status"
WARN_QA:append = " patch-status"
```

But in this case it's trivial to add a patch status to every patch file.

MagneFire, 2025-01-26:
`sturgeon` seems to fail at booting. The kernel image works fine, but it just hangs at the bootlogo.

ADB does not become available, meaning that this may be potentially harder to debug.

beroset, 2025-01-27:
> It will take quite a lot of effort to get all recipes to not include references to the TMPDIR. Hence I'd like to propose actually applying said change in `asteroid.conf`. Though this does risk us not actually fixing these warnings, as we haven't done so previously either.

If, as you propose, we keep it as a warning, it will remain as a visible indicator that more needs to be done.  I think this is a reasonable approach.

MagneFire, 2025-01-31:
Gotten `sturgeon` to boot now.

`systemd` has officially disabled cgroup v1 in systemd version 257 (scarthgap shipped with version 256).
To get `sturgeon` boot one must add the `SYSTEMD_CGROUP_ENABLE_LEGACY_FORCE=1` to the kernel command line.

Also see: https://github.com/systemd/systemd/blob/83bf58f39dbe947c1b4f394667f42a0c557fd94c/NEWS#L148

I'll adjust the kernel command line for every affected watch.

MagneFire, 2025-04-06:
Yocto Walnascar seems to be released. This issue will now keep track of upgrading to walnascar instead of styhead.

MagneFire, 2025-04-14:
Alright, it seems to work now on at least `sturgeon`. Here's what I've tested/fixed:
- Display
  - Launcher
  - AOD
- Sensors
  - Heart rate sensor
  - Tilt-to-Wake
- Vibrator
- Bluetooth
- USB
  - ADB
  - SSH
- Audio
  - Speaker

In the coming days I'll test other platforms.

beroset, 2025-04-14:
I can test `catfish`, `beluga`, `hoki` and `dory` if it helps.

MagneFire, 2025-04-14:
Of course any help is welcome! But don't feel forced to do it either :D

MagneFire, 2025-05-14:
It turns out that libSDL2 has broken at least gpSP.
After some testing it seems that https://github.com/libsdl-org/SDL/commit/0f695adb0392446f36e872e72a828467d5f299f0 broke it resulting in:
```
/usr/src/debug/wayland/1.23.1/src/wayland-client.c:2460: wl_proxy_set_queue: Assertion `proxy->display == queue->display' failed
```

It's fixed in gpSP with this commit: https://github.com/MagneFire/gpsp/commit/d74f472b447ddd646c09c929bd7fd2419f1842f7

MagneFire, 2025-05-17:
For reference this is the plan for testing
- Try varying devices
	- catfish: Recommended watch, should work (Android 9 base) (uses binder interface)
	- sturgeon: Older watch, still good. May get kernel issues as kernel is old (uses non-binder interface, also custom kernel patches for bt)
	- beluga: Modernish watch, kernel is newer
	- hoki: Even more modern
	- emulator: general usage
	- rinato: mainline
- Test the following:
	- Display
	  - Launcher
		  - Do asteroid-* apps launch
		   - Does gpsp (SDL2) launch
	  - AOD
	- Sensors (also check logcat)
	  - Heart rate sensor (app, try twice on sturgeon)
	  - Tilt-to-Wake (sleep display, tilt wrist to activate display)
	- Vibrator
	- Bluetooth
	- USB
	  - ADB (`adb shell`)
	  - SSH (`ssh ceres@192.168.2.15`)
	  - MTP (broken atm)
	- Audio
	  - Speaker
		  - `gst-launch-1.0 filesrc location=input.mp3 ! mpegaudioparse ! mpg123audiodec ! pulsesink`
		   - as `ceres`
	  - Microphone
		   - `gst-launch-1.0 pulsesrc ! audioconvert ! wavenc ! filesink location=recording.wav`
		   - as `ceres`
  - WiFi
  - Gamecontroller (`gpsp`)

Then there's the following remaining:
- Build community packages (✅)
- Build SDK
  - Test SDK

closed 2025-10-06 by beroset