AsteroidOS/asteroid #308: asteroid-health

foscone, 2025-04-13:
Hi, is the asteroid-health project still active?
I tried to clone the [dodoradio](https://git.dodorad.io/dodoradio/asteroid-health) repository but it seems that it cannot be cloned.
I would like to install asteroid-health in my TicWatch Pro (catfish), can anyone tell me how to do it?
Thanks

foscone, 2025-04-13:
Sorry, I didn't realize there is a package manager with all the astroidOS packages.
Connecting via SSH I gave the commands:
```
opkg update
opkg install asteroid-health
```
And I managed to install asteroid-health

closed 2025-04-13 by foscone

JRDetwiler, 2025-04-24:
@foscone Did you do anything else to get this working? It doesn't seem to be in the default repos for me...

```
* opkg_prepare_url_for_install: Couldn't find anything to satisfy 'asteroid-health'.
```

Manually building and installing is also resulting in a white screen, so I'm wondering if the build is failing somewhere and just not showing up in the nightly repos currently?

foscone, 2025-04-24:
No, I didn't do anything else. The version of asteroidOS I used is 1.1-nightly.
To get asteroid-healt to work after installing it, I had to restart the smartwatch.

JRDetwiler, 2025-04-24:
But that's the thing - I'm not even getting as far as `opkg install asteroid-health` working. It just doesn't find the package...