AsteroidOS/asteroid #309: [catfish] Unable to install asteroid-health

JRDetwiler, 2025-04-28:
**Describe the bug**
`opkg` can't find `asteroid-health` in the repositories loaded by default in the OS. I bought a TicWatch Pro 2020 off of someone recently, and have been experiencing this on that watch.

**To reproduce**
Steps to reproduce the behavior:
1. I flashed the watch with the latest nightly build (2025-04-25), "real install".
2. Went through setup, paired it to my phone to sync the time, and corrected the timezone.
3. Established a wifi connection.
4. Ran `opkg update && opkg upgrade`. No updates found.
5. Ran `opkg install asteroid-health`.
6. It can't find the package.

**Expected behavior**
Asteroid Health would be available for this model of watch since it has fitness tracking built in.

**Screenshots**
```
root@catfish:~# opkg update && opkg upgrade
Downloading https://release.asteroidos.org/nightlies/ipk/all/Packages.gz.
Updated source 'uri-all-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/anthias/Packages.gz.
Updated source 'uri-anthias-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/armv7vehf-neon/Packages.gz.
Updated source 'uri-armv7vehf-neon-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/bass/Packages.gz.
Updated source 'uri-bass-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/beluga/Packages.gz.
Updated source 'uri-beluga-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/catfish/Packages.gz.
Updated source 'uri-catfish-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/core2-32/Packages.gz.
Updated source 'uri-core2-32-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/dory/Packages.gz.
Updated source 'uri-dory-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/firefish/Packages.gz.
Updated source 'uri-firefish-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/harmony/Packages.gz.
Updated source 'uri-harmony-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/hoki/Packages.gz.
Updated source 'uri-hoki-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/inharmony/Packages.gz.
Updated source 'uri-inharmony-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/koi/Packages.gz.
Updated source 'uri-koi-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/lenok/Packages.gz.
Updated source 'uri-lenok-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/minnow/Packages.gz.
Updated source 'uri-minnow-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/mooneye/Packages.gz.
Updated source 'uri-mooneye-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/narwhal/Packages.gz.
Updated source 'uri-narwhal-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/nemo/Packages.gz.
Updated source 'uri-nemo-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/pike/Packages.gz.
Updated source 'uri-pike-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/qemux86/Packages.gz.
Updated source 'uri-qemux86-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/ray/Packages.gz.
Updated source 'uri-ray-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/skipjack/Packages.gz.
Updated source 'uri-skipjack-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/smelt/Packages.gz.
Updated source 'uri-smelt-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/sparrow/Packages.gz.
Updated source 'uri-sparrow-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/sprat/Packages.gz.
Updated source 'uri-sprat-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/sturgeon/Packages.gz.
Updated source 'uri-sturgeon-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/swift/Packages.gz.
Updated source 'uri-swift-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/tetra/Packages.gz.
Updated source 'uri-tetra-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/triggerfish/Packages.gz.
Updated source 'uri-triggerfish-0'.
Downloading https://release.asteroidos.org/nightlies/ipk/wren/Packages.gz.
Updated source 'uri-wren-0'.
No packages installed or removed.
root@catfish:~# opkg install asteroid-health
 * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'asteroid-health'.
```

**Device (please complete the following information):**
 - Watch Codename: catfish (TicWatch Pro 2020)
 - AsteroidOS Builddate 2025-04-25
 - Version 1.1-nightly

**Additional context**
I also saw it happening with the prior build too (2025-04-18, I think). However, someone else with either the same model or similar watch managed to get it to work recently as well, which is puzzling: [they wrote about it here](https://github.com/AsteroidOS/asteroid/issues/308). 

Is it possible that the builds started failing at some point and they managed to catch a copy while it was still in the repos?

JRDetwiler, 2025-04-28:
I'll also add that I went to the extent of cloning the [`asteroid-health` repo](https://github.com/AsteroidOS/asteroid-health), building it using the [CMake instructions](https://wiki.asteroidos.org/index.php/Creating_an_Asteroid_app#Example_project), and manually installing it. It succesfully built and installed, but launched into a white screen, which... makes me think that something's not right. Perhaps those sample instructions aren't in line with how the app is compiled with the build pipeline's recipe, but it does seem to come down to the app not working / building properly right now.

Spark2oo7, 2025-05-02:
I have some issue on `beluga` (oppo watch 41mm). I think its problem with `opkg`, not your watch.
I cant install qmltester using `opkg install asteroid-qmltester` from [documantation](https://wiki.asteroidos.org/index.php/Watchface_and_Package_Installation#Installation_of_prebuilt_packages).
If i try run `opkg list | grep asteroid` i got a lot of apps, like `asteroid-calculator`, `asteroid-settings`, `asteroid-weather`. there are also non-standard applications, for example `asteroid-shopper`. I installed it and it works well. But there not other apps like `asteroid-health`, `asteroid-map` and `asteroid-qmltester`.
I had problems when I launched `opkg update` a few days ago, it just downloaded the first (sometimes second) file endlessly. maybe they changed something on the server and accidentally deleted non-standard applications from their repositories?

JRDetwiler, 2025-05-03:
Agreed, it's not just `asteroid-health`, that's just the one I cared about since I primarily wear a watch as a pedometer. Also still seeing the same problem after flashing with the 2025-05-02 build for `catfish`.

Spark2oo7, 2025-05-03:
I check archives and found this:
[2025-04-11](https://release.asteroidos.org/archive/2025-04-11/ipk/armv7vehf-neon/) there all apps (include `asteroid-health`)
[2025-04-18](https://release.asteroidos.org/archive/2025-04-18/ipk/armv7vehf-neon/) there only default apps

So someone install  `asteroid-health` properly (https://github.com/AsteroidOS/asteroid/issues/308#issuecomment-2800015318) 13 april. But now there no many apps in the repositories.

I try download manually .ipk file from archive, but opkg wont install it
```
 * open_outer: Failed to open package 'asteroid-qmltester_+git0+de2672f6da-r1_armv7vehf-neon.ipk': Unrecognized archive format (errno=84)
```
if I'm unpacking an archive on pc and try run executable file use ``invoker --single-instance --type=qtcomponents-qt5 /path/to/qmltester.ipk`` I got errors too:
```
invoker: error: Failed to initiate connect on the socket for type qtcomponents-qt5.
invoker: warning: Booster qtcomponents-qt5 is not available. Falling back to generic.
invoker: error: Failed to initiate connect on the socket for type generic.
invoker: warning: Can't try fall back to generic, already using it
invoker: warning: Connection with launcher process is broken. 
invoker: error: Start application /path/to/qmltester.ipk as a binary executable without launcher...
execve: Exec format error
```

I will try to reconfigure opkg to use [archive](https://release.asteroidos.org/archive/2025-04-11/), but I'm not sure if I can.

Spark2oo7, 2025-05-03:
I got it!

### How to enable archive with opkg
1. connect watch to pc (adb or [ssh](https://wiki.asteroidos.org/index.php/SSH))
2. connect watch to internet [documentaion](https://wiki.asteroidos.org/index.php/IP_Connection)
3. cd to `opkg` configs
```
cd /etc/opkg
```
4. create backup of config
```
cp base-feeds.conf /home/root/opkg-backup
```
5. change repositories to [archive](https://release.asteroidos.org/archive/)
```
sed -i 's#nightlies#archive/2025-04-11#g' base-feeds.conf
```
6. update `opkg`
```
opkg update
```
7. **Now you can install community apps:**
- health:

```

opkg install asteroid-health
```
- map:
```
opkg install asteroid-map
```
- qml tester:

```
opkg install asteroid-qmltester
```
8. maybe restart the watch? (help me with asteroid-health)

JRDetwiler, 2025-05-05:
Thanks for the nice workaround, I was able to install from the archives too.

JRDetwiler, 2025-05-07:
However, the step counting isn't working on a `catfish` (TicWatch Pro 2020), using the `2025-04-11` archive version of `asteroid-health`. Feels like recent builds could be failing for a reason.