AsteroidOS/asteroid #311: Migrate to Qt6

beroset, 2025-05-08:
Qt5.15 will be EOL (end of life) on 26 May 2025.  We should move to Qt6.8 LTS as soon as possible.

PureTryOut, 2025-05-08:
For context, https://www.qt.io/blog/extended-security-maintenance-for-qt-5.15-begins-may-2025

PureTryOut, 2026-03-20:
That previously mentioned PR seems to have gone stale and requires a bunch of work to work again. I took large parts of it in https://github.com/AsteroidOS/qml-asteroid/pull/83 and updated it to work with the current release of qml-asteroid. It also allows installation of both a Qt5 and Qt6 version alongside each other which should make the transition easier. Instead of porting everything at once we can this way port application per application, working up to the shell itself, and start testing them while the transition is happening

See https://github.com/AsteroidOS/qml-asteroid/pull/83 for Qt6 support in qml-asteroid and https://github.com/AsteroidOS/asteroid-calculator/pull/12 for the first app that makes use of that. We can move on from there once that's merged.